Gi-z/CSIKit #2: RSS RSSI dBm dB

tweigel-dev, 2020-12-16:
Hay @Gi-z ,
I am very interested extracting csi with the intel5300. I noticed that you are using some snippets of the python PR  by @pgawlowicz.
I didn't understand at his code, what the difference of rssi_db and rssi is because at rssi he doese some stuff with magic number 44 and the agc. 
```
rssiA_db = rssiA - 44 - agc
```
Did you know that the 44 is ?
which roll does the agc play at the rssi value ?
The results are negativ at rssi_db .  
what did you think is the unit of the rssiA value we are collecting directly from the udp-package?
In my eyes the definition of RSS and RSSI are unclear at the wifi usage.

Gi-z, 2020-12-17:
Hi, so my code is actually based on the matlab examples included with the Linux 802.11 CSI Tool Supplementary package (https://dhalperi.github.io/linux-80211n-csitool/faq.html). As far as I'm aware, Halperin included this fixed offset to account for internal scaling by the Intel firmware. 

I would be unable to vouch for the accuracy of my implementation, but I trust their work over my own in that regard. In verifying this code is accurate I have used the same test file as included with the Supplementary package and generated matlab matrices from numpy matrices. I have them compared them to the output from given matlab code and found them to be identical (minus insignificant rounding errors).

Negative dBm is as it both a relative magnitude and on a logarithmic scale. [This](https://www.reddit.com/r/explainlikeimfive/comments/88wy9d/comment/dwnv2s3) Reddit answer outlines the rough idea, however I would suggest googling more about negative dB as that may not be the best source!

tweigel-dev, 2020-12-17:
Thanks for your resonse!
I am aware of what the difference beween dBm and dB is. Tanks for the Link! there is described what I asked for here ^^ 

> rssi_a, rssi_b, and rssi_c correspond to RSSI measured by the receiving NIC at the input to each antenna port. This measurement is made during the packet preamble. This value is in dB relative to an internal reference; to get the received signal strength in dBm we must combine it with the Automatic Gain Control (AGC) setting (agc) in dB and also subtract off a magic constant. This process is explained below.

closed 2020-12-17 by tweigel-dev