Gi-z/CSIKit #5: generates correct Phase Difference

helloworld1973, 2020-12-29:
Thanks a lot for your work.
1. I am new in CSI and saw your code here, is there any clearer calculation on phase?

metric == "phasediff":
                if scaled_entry.shape[1] >= 2:
                    #Not 100% sure this generates correct Phase Difference.

2. I know the complex number is (gain, phase). how to transfer it to amplitude? why you use 20*log(gain)?

Gi-z, 2020-12-30:
Hi, thanks for the issue.

1. I'm not particularly knowledgeable about phase difference calculation, however this is a valid method to calculate phase difference using the phase values from two different receiving antennas.  The problem with this is that the phase values generated this way have an unknown (random) offset applied to each one. There are a few methods for getting around this. I would suggest reading [this](https://par.nsf.gov/servlets/purl/10055601) paper for a thorough analysis on the subject with context.

2. CSI amplitude is typically measured in dB/dBm. Daniel Halperin's examples from their Linux CSI Tool (supplementary code) use the `db` function from Matlab. Using 20*log(gain), we can replicate this behaviour. Note that this conversion can be used for either voltage or power, and the calculation used as such varies. As we're converting for amplitude, the power variant should be used.

Hope this clears up any questions. Happy to point to any other useful sources.

tweigel-dev, 2021-01-26:
Hay @helloworld1973 ,
you can calculate the phase by using cmath.phase(complex) 
or you can calculate the amplitude by using abs(complex)
This is valid because the csi is assembled like that:
[image]

closed 2021-03-25 by Gi-z