Gi-z/CSIKit #7: CSIKIT For atheros

serrhini, 2021-03-23:
The csikit return always an error with atheros data see please:
C:\CSIKit\CSIKit-master\CSIKit\data\atheros>csikit example_bigEndian.dat
Invalid code for beamforming measurement at 0x3.
Traceback (most recent call last):
  File "C:\Users\userfso\AppData\Local\Programs\Python\Python39\Scripts\CSIKit-script.py", line 33, in <module>
    sys.exit(load_entry_point('CSIKit==0.0.21', 'console_scripts', 'CSIKit')())
  File "c:\users\userfso\appdata\local\programs\python\python39\lib\site-packages\CSIKit\__main__.py", line 35, in main
    display_info(args.file)
  File "c:\users\userfso\appdata\local\programs\python\python39\lib\site-packages\CSIKit\get_info.py", line 5, in display_info
    reader = get_reader(path)
  File "c:\users\userfso\appdata\local\programs\python\python39\lib\site-packages\CSIKit\reader.py", line 12, in get_reader
    return IWLBeamformReader(path, scaled=True)
  File "c:\users\userfso\appdata\local\programs\python\python39\lib\site-packages\CSIKit\read_bfee.py", line 36, in __init__
    self.csi_trace = scale_timestamps(self.csi_trace)
  File "c:\users\userfso\appdata\local\programs\python\python39\lib\site-packages\CSIKit\csitools.py", line 138, in scale_timestamps
    csi_trace[0]["timestamp"] = 0
IndexError: list index out of range

can you please enlight me

Gi-z, 2021-03-24:
Hi, this was caused by a bug with an older version of CSIKit. If you install the latest version (1.0.8), you should have no issue using CSIKit with the included example files.

`pip install --upgrade csikit`

Some usage of the framework has changed since 0.0.21, so I would recommend taking a look at the updated README for examples. Let me know if you have any other issues!

serrhini, 2021-03-25:
Thank you, Yes I will return to you for any issues, I'm new in this area and find this project very helpful for starting.
Regards,

closed 2021-03-25 by Gi-z