Gi-z/CSIKit #11: IndexError: list index out of range

makoc0704, 2021-06-16:
Hey, 

I have collected data using the Nexmon-CSI-Tool: 
https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51-plus

I've collected the CSI data and store them in a pcap file :
[output.zip](https://github.com/Gi-z/CSIKit/files/6665286/output.zip)

I dont even get an output of these File:
`csikit output.pcap`
`Traceback (most recent call last):`
 `
 File "/usr/local/bin/csikit", line 10, in <module>
    sys.exit(main())
`
 `
 File "/usr/local/lib/python3.7/dist-packages/CSIKit/__main__.py", line 57, in main
    display_info(args.file)
`
  `
File "/usr/local/lib/python3.7/dist-packages/CSIKit/tools/get_info.py", line 7, in display_info
    metadata = csi_data.get_metadata()
`
  `
File "/usr/local/lib/python3.7/dist-packages/CSIKit/csi/csidata.py", line 45, in get_metadata
    final_timestamp = timestamps[-1]
`
`IndexError: list index out of range`

If i use your examples, it works. 
Thanks in advance!

Gi-z, 2021-06-17:
Hi, this appears to be a parsing issue. The chipset header in your CSI data appears to be slightly different to those I've previously observed, and while the file does not successfully parse under the bcm43455c0 (Pi 3B+/4) format, it successfully parses with the bcm4366c0 (Asus RT-AC86U) format.

Can you confirm the hardware you used to collect this data? e.g. Raspberry Pi 4, Asus RT-AC86U, or another device. Thanks.

makoc0704, 2021-06-17:
Thank you for your prompt reply!
I have set up a Raspberry Pi 4 Model B Rev 1.2 as AD and ping it with my laptop.
With another Raspberry Pi 4 Model B Rev 1.2 - on which the nexmon software is installed - I collect the CSI data as described in the documentation. 

Thank you for your help!

makoc0704, 2021-06-17:
Sure!
**makecsiparams settings:**
`mcp -c 36/80 -C 1 -N 1 -m <MAC of Pi - Access Point, that receive ping>` 
` KuABEQAAAQDcpjJNqg8AAAAAAAAAAAAAAAAAAAAAAAAAAA==`
`ifconfig wlan0 up`
`nexutil -Iwlan0 -s500 -b -l34 -vKuABEQAAAQDcpjJNqg8AAAAAAAAAAAAAAAAAAAAAAAAAAA==`
`iw dev wlan0 interface add mon0 type monitor`
`ip link set mon0 up`
`tcpdump: listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes`
`Got 31`

**nexutil:** 
`nexutil -k`
`chanspec: 0xd022, 34`

If I enter `nexutil -k 36/80` I get an error: 
```
nexutil -k 36/80
nexutil: Too many arguments
```

Gi-z, 2021-06-17:
The format for setting a channel with nexutil is `nexutil -k36/80`. Not sure why spaces aren't allowed 🤷 

This is interesting though. I have been able to get your file to parse and this seems to be an exacerbated version of an issue I've encountered before. The pcap frame header appears to be 26 bytes larger than normal. I'm implementing a solution for this and am hoping to have it ready by the end of the day. I've included a visualisation of your data below as proof!

[image]

makoc0704, 2021-06-17:
Great, thank you very much for your help!

Gi-z, 2021-06-17:
Apologies for the delay. This'll take a little time to implement elegantly.

In the interest of making sure you have something, I've linked a temporary build you can install which will parse your files correctly. Any other nexmon filetypes will fail, but this will work with the system you're currently running.
https://drive.google.com/file/d/1rjnCiJgeh7CZFN_ZfQEWGfcv1dZBp-Qt/view?usp=sharing

You can install it by navigating to the root folder (containing setup.py) and running: 
```console
pip uninstall csikit
python setup.py install
```

I hope to have a solution for this issue merged into master soon. Best of luck with your work!

makoc0704, 2021-06-18:
Awesome, it works! 
Big thanks for your help :-)

closed 2021-06-18 by makoc0704