Gi-z/CSIKit #12: Confusing about the method of reversing the AGC

GuoJingtao-1997, 2021-06-17:
Hi Glenn,

I'm quite appreciate for this great project. But I'm still confused about something that shown in this issue:https://github.com/Gi-z/CSIKit/issues/9#issue-861477838.

First, why it requires a sqrt computation for the formula that is used for reversing the AGC? Is it due to the energy that we used to compute for this formula? As I this both components in this formula have the same unit, therefore, we may not need to do a sqrt computation.

Second, I saw that you multiply csi data with its conjecture and I think the result after doing this will only be a real value rather than a complex value. I think we may multiply csi data with itself. Am I missing something? Thank you so much in advance.

Best Regards

Gi-z, 2021-06-17:
Hi, sorry for the late reply.

First: Admittedly I did not fully understand the AGC reversal process when developing a solution. I am not an engineer, so this approach may not yield perfect scaling. You are correct in that both components have the same unit and it may be the case that the sqrt operation is not required. I'm preparing some experiments to further assess the efficacy of this approach to AGC reversal, and this approach was largely focussed on addressing intra-frame scaling issues as opposed to overall value scaling. Removing the np.sqrt operation in the example I provided in #9 does yield what I would consider to be a better overall scale for the data.

I'm happy to leave this issue open as a space to further discuss this approach and how it can be improved. A device-agnostic approach to AGC reversal for CSI using RSS is beneficial to the entire research community.

Second: Much easier one to answer. You are correct in that there will only be a real value after the operation, however numpy does not change the dtype of the array afterwards.

[image]

You can see after multiplying the array by it's complex conjugate, all imaginary components are 0, but they are still present! By selecting the real values with np.real, we get a float64 array instead.
```python
>>> np.real(test).dtype
dtype('float64')
```

GuoJingtao-1997, 2021-06-18:
That's no problem. Thank you so much for your reply : )

I have another question that why you multiply csi data with its complex conjugate rather than the data itself. Since I saw that the formula is csi^2, so could we multiply csi data with itself and then take the real value? Thank you so much.

Gi-z, 2021-06-18:
It is not simply the square of the matrix we are calculating, but the absolute square. By multiplying the matrix by its complex conjugate, we can perform both operations (np.abs(x), and np.power(x, 2)) at once. The result of said operations is mostly similar however and I suppose either method is fine.

'''python
>>> import numpy as np
>>> x = np.array([1+2j, 3+5j, 8+9j])
>>> np.abs(np.power(x, 2))
array([  5.,  34., 145.])
>>> np.multiply(x, np.conj(x))
array([  5.+0.j,  34.+0.j, 145.+0.j])
'''

GuoJingtao-1997, 2021-06-18:
Cool! Thank you so much for your help and really appreciate for such great project.

closed 2021-06-18 by Gi-z