Gi-z/CSIKit #14: spatial stream index of nexmon daa

yujianyuanhaha, 2021-06-28:
greetings,

Thanks for the good work!

@Gi-z it looks like the spatial stream index is NOT implement, do you have any clue for that?

It is pretty easy, in matlab it is like 
```
payloadbytes = typecast(payload,'uint8');
if strcmp(DEVICE,'ASUS')
    
    core = bitand(payloadbytes(56),3); %
    rxss = bitand(bitshift(payloadbytes(56),-3),3);
    Nss(k) = core*4 + rxss;
    
else
    % payloadbytes(55)
    % 0 8 16 24
   % for NEXUS phone and PI
    Nss(k) =  round( payloadbytes(55)/8);
end
```
right after line 36 at [csireader.m](https://github.com/seemoo-lab/nexmon_csi/blob/master/utils/matlab/csireader.m)

Best

Jet

Gi-z, 2021-06-28:
Hi, I hadn't implemented this feature yet since I needed a file with an example of multiple spatial streams. If you have a file you could provide as an example, I'd love to get that implemented soon!

yujianyuanhaha, 2021-06-28:
@Gi-z here you go, this is a 4x4 spatial stream [ASUS_example.pcap](https://drive.google.com/file/d/1jCQlT49_fZxDSQqgfgpFUkk2AfMJvIrS/view?usp=sharing)

Gi-z, 2021-06-28:
Thanks! The core/spatial bytes are currently parsed in each frame. I'll just need to update the parsing logic to interpret antenna index correctly, and `csitools.getCSI` to assemble matrices considering antenna index in sequential frames.

Gi-z, 2021-07-01:
Hi, I've got this implemented however it's highlighted how slow my `unpack_float_acphy` implementation is. This'll likely need to be resolved before I finish adding this feature.
 
Example, first 4 cores on the file you supplied:

[image]
[image]
[image]
[image]

Would you be ok with me including this in the library .as an example of BCM4366c0 data from a 4x4 antenna array?

yujianyuanhaha, 2021-07-01:
@Gi-z Sure, thanks again for fixing this.

yujianyuanhaha, 2021-07-02:
@Gi-z another way to accelerate the computing is do the downsample, i.e only do one CSI reading each 100 packages, it is pretty easy in matlab, change line 27 of [csireader.m](https://github.com/seemoo-lab/nexmon_csi/blob/master/utils/matlab/csireader.m) from
```
f = p.next();
```
to 
```
 for l = 1:100     % downsample by 100
        p.next();
end
f = p.next();
```

Gi-z, 2021-07-24:
Apologies for the delay. I was on a 3-week holiday, and have now had time to improve the performance. This feature is now implemented in v2.2.0. If you have any issues using this feature, don't hesitate to let me know and I'll get it sorted asap. Thanks for your help!

closed 2021-07-24 by Gi-z

t5670116, 2022-07-06:
@yujianyuanhaha 
I have looked at your pcap file and one out of every four values is 0.
Why is that? I would be happy to know.
I also want to get the value of the 4x4 CSI in ASUS but the value is 0 when the cores are 3 and 4.
I want to get the value of th
[image]
e 4 x 4 CSI. I need your help.