Gi-z/CSIKit #19: The PicoScenes platform now supports 802.11ac/ax CSI extraction by Intel AC9260/AX200 Wi-Fi NICs

jiangzp, 2021-10-04:
I am the author of the PicoScenes platform. The PicoScenes platform has just experienced a major update. It is now the first and currently the only platform that supports CSI extraction for 802.11ax-format frames. More information is at ps.zpj.io.

Your CSIKit is great. If you need assistance, just contact me.

Gi-z, 2021-10-04:
Thank you, I'm a huge fan of PicoScenes and have used it for a few recent experiments.
I'd love to finish adding support for the mvm format, however I'm experiencing issues with my own AX200 setup. I'll make another issue so it's accounted for properly.

jiangzp, 2021-10-05:
Wow~~ I didn’t realize you are Gleen 😃. So great you made this tool👍👍

jiangzp, 2021-10-05:
Thanks for your filing the AX200 firmware bug issue. I have replied in PicoScenes Issues Tracker. It is a confirmed bug about using Intel’s latest unpublished firmware. I will fix it ASAP.

jiangzp, 2021-10-05:
Regarding the PicoScenes’s complex RxFrame data structure, I believe the easier and more consistent approach to support PicoScenes format is to use C++ to parse the binaries and the en translated the resulted high-level structure into Python.  You know the fundamental XXXSegment structure is versioned and subjected to change due to the future update. 

Actually, we are also creating the official version of PicoScenes Python Toolbox. Maybe you can wait a little while allowing us to perfect it and release. Building atop the official Python conversion can save you a lot lot of energy😂.

Gi-z, 2021-10-05:
Thanks for your explanation of the issue and I look forward to a resolution.

I agree that raw Python parsing is not the most efficient method for dealing with the binary data structures. I've seen alternative libraries which use Cython, etc, however I'm trying (whether successful or not) to ensure the code is still quite easy to interpret for anyone who might be developing their own solutions using my code as a base. That said, when performance becomes too poor with raw Python approaches (as is starting to become the case with `nexmon_csi` 4x4 MIMO 80MHz data) I may have to reconsider my options.

For my current experiments, I'm happy with being able to use CSIKit to parse data produced with the SDR frontend in PicoScenes with the USRP B210 I'm using. If anyone else can get use out of it, that is also good. When the PicoScenes Python Toolbox becomes available, I would love to look at replacing the parsing backend to build upon the official toolbox 😄.

closed 2022-06-15 by Gi-z