Gi-z/CSIKit #20: Ambiguous Wording in NEXCSIFrame section

CDeLeon94, 2021-10-04:
https://github.com/Gi-z/CSIKit/blame/master/README.md#L191

Not clear that the link on `mzakharo` is to a PR that supports the RSSI measurement. 

As a result I ended up installing the `modified version of [nexmon-csi](https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51)` which doesn't support RSSI extraction.

Gi-z, 2021-10-04:
Thanks for picking this up. The README is long overdue an overhaul and I'm aiming to move things over to a more comprehensive documentation system like sphinx.

Apologies for the issue, I know all too well how long it takes to set up `nexmon_csi` even in the best of circumstances. I'll get the wording there updated and will keep it in mind when writing new docs. Hopefully [T3rO's](https://github.com/T3rO/nexmon_csi_gain) AGC recovery method gets merged in some form in the near future!

Gi-z, 2021-10-04:
Hopefully that update clears things up a bit. It's my hope that a unified version of the `nexmon_csi` format with all of these features will become available in the near future!

CDeLeon94, 2021-10-04:
Yeah, that's a little more clear. 

I ended up installing zeroby0's fork, that installer script is much much easier. 
Then I noticed their [pi-5.4.51-plus](https://github.com/nexmonster/nexmon_csi/tree/pi-5.4.51-plus) branch that has support for RSSI and even bakes in some blank space for future data extraction without requiring exiting data to be shifted. 

Might just modify their [python](https://github.com/nexmonster/nexmon_csi/tree/feature/python) branch to decode the `plus` variant of the data. 

Glad to see there are so many people working in this space, though the many active forks and branches is a little overwhelming at times.

Gi-z, 2021-10-04:
Big fan of Aravind's work for sure. Would recommend finding a solution that fits your area of work best and sticking with it. Constantly changing devices/configurations is a nightmare.

Would suggest taking a look at the scaling method here: https://github.com/Gi-z/CSIKit/blob/master/CSIKit/util/csitools.py#L74
It's based on the method proposed in CRISLoc: https://doi.org/10.1109/JIOT.2020.3022573
I'm hoping to have some work published soon which highlights upsides and downsides to using this approach.

Best of luck in your work, and feel free to let me know if I've missed any other similar issues!

closed 2021-10-04 by Gi-z