Gi-z/CSIKit #24: extract the amplitude and phase information of csi from CSV file

zhy102, 2022-03-05:
Thanks for the code, but I only have the CSV file with csi obtained from esp32, I want to extract the amplitude and phase information from it, I can't see it is complex at all from the matrix of CSI_Data, how should I extract them? I'm not familiar with python, I'm good with matlab. thanks!

Gi-z, 2022-03-05:
Hi, so just like the example, you can load the your CSI data from the ESP32 like so:

`
from CSIKit.reader import CSVBeamformReader
from CSIKit.utils import csitools

reader = CSVBeamformReader()
data = reader.read_file("path/to/your/esp32/csv")
csi_matrix, no_frames, no_subcarriers = csitools.get_CSI(data, metric="amplitude or phase")
`

This will generate your CSI matrix and extract either amplitude or phase depending on the metric you select.
Let me know how that works out for you!

zhy102, 2022-03-05:
OK， thanks！

Gi-z, 2022-03-05:
Oh, and you can also extract the raw complex values by setting `metric=""`.

zhy102, 2022-03-05:
OK， thank you so much!

closed 2022-03-10 by Gi-z