Gi-z/CSIKit #25: Visualisation is in dBm

Seppe-d, 2022-03-30:
When using CSIkit from the command line, the graph has dBm as a unit. When I use CSIkit as a library, I get results in the same order of magnitude. Could it be that the CSIkit library does the transformation to dBm while reading in a file? Or does the Atheros CSI tool put out CSI in dBm form?

Gi-z, 2022-03-30:
The `--graph` option plots CSI amplitude in dBm.

CSI files (like those produced by the Atheros CSI Tool) contain raw CSI, complex pairs of values representing gain and phase measured at each subcarrier. Once CSIKit reads this data in, your `CSIData` object (returned by `read_file`) will contain a `frames` attribute containing each frame of CSI and its `csi_matrix`. This is still in the raw form, which has no units. 

The `csitools.get_CSI` function is used to pull these individual matrices together into one CSI matrix. There are several options in this function to transform your CSI. `metric="amplitude"` will return CSI amplitude, `metric="phase"` will return CSI phase, and `metric=None` will return the raw complex values. The `extract_as_dbm` option is set to True by default, but you can set it to False if you want your amplitude back in mW.

Hopefully that clears it up, but let me know if anything doesn't seem right.

Seppe-d, 2022-03-30:
Aha, i indeed want the raw complex values. You just solved my problem, thanks!

Gi-z, 2022-03-30:
Glad to hear it! I'll close this issue, but let me know if you have any other questions.

closed 2022-03-30 by Gi-z