Gi-z/CSIKit #26: IWL5300 IndexError: list index out of range

bachejames, 2022-04-13:
Hi there!

Thanks so much for putting together this super cool tool, huge fan!!

I'm trying to use CSIKit to parse `.dat` files from IWL5300, however I'm constantly getting the `IndexError: list index out of range` error. I noticed this error came up once before, however I think that was resolved with a new version of CSIKit, and I am using the latest version of CSIKit (2.3).

### My attempt
Example sample of files I have tried: [example_1.dat](https://drive.google.com/file/d/18trTF_WnLoWT51cLu2n6-Y-HxkAWZAVh/view?usp=sharing), [example_2.dat](https://drive.google.com/file/d/19_bL0uSb-7KVbiKqBUF_nS1DS_C8w3vv/view?usp=sharing) or [example_3.dat](https://drive.google.com/file/d/1hhWtBQo7g_6XqQR0-labkHtudoDNSbIn/view?usp=sharing)

Running `csikit file_here.dat`, returns:

```
james@james ~ % csikit file_here.dat 
Traceback (most recent call last):
  File "/usr/local/bin/csikit", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/site-packages/CSIKit/__main__.py", line 57, in main
    display_info(args.file)
  File "/usr/local/lib/python3.9/site-packages/CSIKit/tools/get_info.py", line 8, in display_info
    metadata = csi_data.get_metadata()
  File "/usr/local/lib/python3.9/site-packages/CSIKit/csi/csidata.py", line 38, in get_metadata
    unmodified_csi_matrix = self.frames[0].csi_matrix
IndexError: list index out of range
```

Any suggestions you can offer would be hugely appreciated - thank you!

Gi-z, 2022-04-13:
Hi, I can definitely see the issue you're having. Took a quick stab at trying to fix it, but there seems to be some misalignment in the underlying structs. This particular error occurs because the `expected_length` for each frame's CSI payload doesn't match the actual length of the buffer passed to that function. It's likely a single byte misalignment.

Out of interest, how did you produce these files? (Device configuration, AP/STA, etc)

Am currently focused on my thesis, will aim to fix this when I can.

bachejames, 2022-04-13:
Thanks so much for your help @Gi-z, hugely appreciate it, particularly during your thesis - best of luck with it! That's a fascinating insight.

Sure thing:
AP: One Intel Galileo, using an IWL5300, with one antenna attached
STA: Three Intel Galileos, each using an IWL5300, with one antenna attached to each board

Thank you!!