Gi-z/CSIKit #27: Intel IWL5300   IndexError: index 2 is out of bounds for axis 0 with size 2

ljl199711129, 2022-04-20:
Thank you for the CSIKit tool library, but I'm having some problems reading the data.
(1)
[image]
I am using a public dataset, antenna configuration: 3 Rx, 3 Tx,
(2)At the same time, I get the same error as the comment
[image]
Hope to get your answer, thank you

Gi-z, 2022-04-20:
Hi, could you provide an example of one of the files that fails to parse?

ljl199711129, 2022-04-20:
[https://data.4tu.nl/articles/dataset/Channel_state_information_WiFi_traces_for_6_activities/12692816/1](url)
1_falling_2.dat in folder day_1 belongs to error 1 ( IndexError: index 2 is out of bounds for axis 0 with size 2 );
1_walking_5.dat in folder day_1 belongs to error 2 ( IndexError:list index out of range)
Thank you for your prompt reply.

Gi-z, 2022-04-20:
> https://data.4tu.nl/articles/dataset/Channel_state_information_WiFi_traces_for_6_activities/12692816/1
> 1_falling_2.dat in folder day_1 belongs to error 1 ( IndexError: index 2 is out of bounds for [...]

Hi, I just tried with my `personal_build` branch. Both files parse correctly with no issues. Could you try installing it and checking it's working for you?

You can install the specific branch with the following command:
`pip install git+https://github.com/Gi-z/CSIKit.git@personal_build`

Gi-z, 2022-06-15:
This issue has been resolved in the current release (v2.4).

closed 2022-06-15 by Gi-z