Gi-z/CSIKit #28: Any way to get real-world time from CSI data?

dev-sora, 2022-05-11:
Hi,

I'm seeking the way to get real time from CSI data observed by Intel AX210.
Could someone tell me how to do it?

dev-sora, 2022-05-11:
@Gi-z 
Thanks for your reply, and I have to say sorry for any misunderstanding my question may have caused you.
My question is that I would like to know how to convert the timestamps I can get from PicoScenes to real time(ex: HH:MM:SS.f). The timestamps I get are apparently not UNIX time, so I don't know how to convert them to real time.

Gi-z, 2022-05-11:
Ah right, my apologies, should've read more thoroughly.

Intel's AX-series timestamps in PicoScenes are recorded by the internal clock on the NIC. Real-world timestamps aren't currently available through PicoScenes, as I'm aware. The same is the case for the IWL5300, and many other COTS CSI devices. Your best bet would be to take the first timestamp as your file creation time, and use the difference between timestamps to establish the interval for each measurement.

closed 2022-12-03 by Gi-z

ElieCaratge, 2023-02-15:
Hello @dev-sora , what you tried to di is exactly what i'd like to reproduce. Could you tell me more about the hardware setup you used ?
Thank you very much !