Gi-z/CSIKit #29: Rasberrypi 4(bcm43455c0) IndexError: list index out of range

Cheon932354, 2022-07-26:
Thank you for your awesome tools.
i am a studying indoor localization using CSI

(CSIKit) C:\git\CSIKit-master>csikit -c output2.pcap
Traceback (most recent call last):
  File "C:\Anaconda\envs\CSIKit\lib\runpy.py", line 192, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Anaconda\envs\CSIKit\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Anaconda\envs\CSIKit\Scripts\csikit.exe\__main__.py", line 7, in <module>
  File "C:\Anaconda\envs\CSIKit\lib\site-packages\CSIKit\__main__.py", line 47, in main
    generate_csv(args.file, args.csv_dest, args.csv_metric)
  File "C:\Anaconda\envs\CSIKit\lib\site-packages\CSIKit\tools\convert_csv.py", line 8, in generate_csv
    csi_data = reader.read_file(path)
  File "C:\Anaconda\envs\CSIKit\lib\site-packages\CSIKit\reader\readers\read_pcap.py", line 284, in read_file
    data_frames = self.read_frames(self.pcap.frames, scaled, ret_data.bandwidth)
  File "C:\Anaconda\envs\CSIKit\lib\site-packages\CSIKit\reader\readers\read_pcap.py", line 417, in read_frames
    if frames[0].payloadHeader["sequence_no"] == frames[-1].payloadHeader["sequence_no"]:
IndexError: list index out of range

i want to convert .pcap to .csv
So I used the data that I have.
I collected CSI through Raspberry Pi4 using Nexmon_csi.(1000 CSI)

How can i solve this problem?
i hope you give me an answer.
thank you

Gi-z, 2022-07-26:
Hi, can you please upload the file you are trying to parse? With it, I can debug the issue.

Cheon932354, 2022-07-26:
thank you for quick answer[](url
[output2.zip](https://github.com/Gi-z/CSIKit/files/9187788/output2.zip)
)

this is my file

Gi-z, 2022-07-26:
> thank you for quick answer[](url [output2.zip](https://github.com/Gi-z/CSIKit/files/9187788/output2.zip) )
> 
> this is my file

I've just checked the file you sent, and it fails to parse because there is no CSI data from nexmon_csi in there. nexmon_csi generates UDP packets which contain the CSI data within the payload. Instead, the capture file contains TCP traffic between your Pi and another device.

[image]

nexmon_csi's documentation lists the example command (`tcpdump -i wlan0 dst port 5500`) for capturing CSI data. This last part is important, as it sets a port filter in tcpdump so non-CSI traffic is not included in the capture. To write the data to a file you can append `-w my_filename.pcap`, and if you wish to see a realtime readout of the packet counts you can also append `-vvv`.

Cheon932354, 2022-07-26:
thank you for your anwser~
i want to capture packet between my pi and another device.
but i have another issue about nexmon_csi

`root@raspberrypi:/home/pi/nexmon/utilities/nexutil# tcpdump -i wlan0 dst port 5500 -vv -w output0-1.pcap -c 1000
tcpdump: listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
^C0 packets captured
0 packets received by filter
0 packets dropped by kernel`

nexmon_csi can't capture packet...

Cheon932354, 2022-07-26:
hello

Sorry to bother you.
This time, i used UDP as input.
But the same issue happens, so why?
This is a collection of UDP.The pcap file
[output.zip](https://github.com/Gi-z/CSIKit/files/9191423/output.zip)
.

`(CSIKit) C:\git\CSIKit-master>csikit -c output.pcap
Traceback (most recent call last):
  File "C:\Anaconda\envs\CSIKit\lib\runpy.py", line 192, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Anaconda\envs\CSIKit\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Anaconda\envs\CSIKit\Scripts\csikit.exe\__main__.py", line 7, in <module>
  File "C:\Anaconda\envs\CSIKit\lib\site-packages\CSIKit\__main__.py", line 47, in main
    generate_csv(args.file, args.csv_dest, args.csv_metric)
  File "C:\Anaconda\envs\CSIKit\lib\site-packages\CSIKit\tools\convert_csv.py", line 8, in generate_csv
    csi_data = reader.read_file(path)
  File "C:\Anaconda\envs\CSIKit\lib\site-packages\CSIKit\reader\readers\read_pcap.py", line 284, in read_file
    data_frames = self.read_frames(self.pcap.frames, scaled, ret_data.bandwidth)
  File "C:\Anaconda\envs\CSIKit\lib\site-packages\CSIKit\reader\readers\read_pcap.py", line 417, in read_frames
    if frames[0].payloadHeader["sequence_no"] == frames[-1].payloadHeader["sequence_no"]:
IndexError: list index out of range`

Gi-z, 2022-07-26:
The traffic in your most recent file still isn't nexmon_csi traffic. It is UDP traffic, but it is not on port 5500.

You receive no packets in your tcpdump capture. This is because the Pi is not configured correctly.
You need to configure the device properly using `makecsiparams`, as listed in the documentation.

For instance, if you wanted to capture traffic on 2.4GHz Channel 1 with a 20MHz bandwidth, you would configure it this way.
`makecsiparams -c 1/20 -C 1 -N 1`

Your issue is with using [nexmon_csi](https://github.com/seemoo-lab/nexmon_csi) itself, so your best bet is to make an issue [there](https://github.com/seemoo-lab/nexmon_csi/issues/new). I'd be happy to comment on it and continue assisting. This problem is a really common one with a couple of possible causes. The community will be able to assist you and get this running in no time!

closed 2022-07-26 by Gi-z