Gi-z/CSIKit #32: does not read entire length of packets

nabeelni, 2022-09-15:
doesnt read entire length of packets of file on broadcom asus router

nabeelni, 2022-09-15:
if i have 1000 frames lets say it doesnt return all the frames captured from asus router e.g at 40 MHz

Gi-z, 2022-10-18:
CSIKit performs grouping to generate CSI matrices with dimensions of (nRx, nTx, nSub) (Rx=Receiving antennas, Tx=Transmit antennas, Sub=Subcarriers).

For instance, if there are 1000 frames in a .pcap file collected from the ASUS RT-AC86U when operating in 4x4 MIMO mode, then each 16 sequential frames will be from a different Tx/Rx pairing. Those sequential frames contain the matrix from each antenna pairing but represent the CSI measurement for the same 802.11 frame. You can confirm this by inspecting the sequence number associated with each frame. Let me know if that matches up and explains your issue.

Gi-z, 2022-12-03:
Closing this due to inactivity. Feel free to reopen if you wish to continue. 👍

sarielkk, 2023-12-29:
> CSIKit performs grouping to generate CSI matrices with dimensions of (nRx, nTx, nSub) (Rx=Receiving antennas, Tx=Transmit antennas, Sub=Subcarriers).
>
> For instance, if there are 1000 frames in a [...]

Hello, your project helped me a lot. 
I still don’t quite understand why frame count obtained by CSIKIT is different from the UDP number captured using nexmon, and it does not show regularity.

Gi-z, 2023-12-30:
> Hello！ Thank you for your project. It helped me a lot. 
> 
> I still don’t quite understand why frame count obtained by CSIKIT is different from the UDP number captured using nexmon, and it does not show regularity. 

Hi. So the PCAPs you get when using Nexmon contain UDP packets. So when you have a Raspberry Pi with CSI in a PCAP, 600 packets means 600 frames of CSI. This matches the frame count you would observe in the Nexmon MATLAB example. 

However, when you use a device that has multiple antennas (like the ASUS router), CSIKit and the Nexmon MATLAB example behave differently. Each frame in the Nexmon PCAPs contains a marker to indicate which antenna received it. This means that multiple antennas can receive the same frame. As a result, one frame will show up as several frames in the Nexmon MATLAB example, with the total count being nRx * nTx * nFrames = frameCount. CSIKit automatically assembles this data into matrices, where a single frame will have a `csi_matrix` attribute containing an nRx * nTx-dimensional matrix. This means when you use the `get_CSI` function, you get a single matrix of dimensions (nRx * nTx * nFrames, nFrames = frameCount).

Hope this helps, let me know if that clears it up for you.

sarielkk, 2024-01-11:
> Hi. So the PCAPs you get when using Nexmon contain UDP packets. So when you have a Raspberry Pi with CSI in a PCAP, 600 packets means 600 frames of CSI. This matches the frame count you would [...]

Thanks for your reply, I understand the problem.
I also found another problem when using CSIKIT. When I processed the data collected by the single-antenna transmitter (router) and the dual-antenna receiver (Nexus 6P), CSIKIT showed that my data was 1Rx, 2TX. When I processed the data collected by the dual-antenna transmitter (router) and the single-antenna receiver (Nexus 6P), CSIKIT shows that my data is 2Rx, 1Tx, but there is only one valid set of data when viewing the csv file. May I ask why this happens?
Finally, I attach the data file I collected using Nexus 6P.
c3n1.pcap is Tx single antenna, Rx dual antenna configuration;
c1n3.pcap is the configuration of Tx single antenna, Rx dual antenna
[c1n3.pcap.gz](https://github.com/Gi-z/CSIKit/files/13896667/c1n3.pcap.gz)
[c3n1.pcap.gz](https://github.com/Gi-z/CSIKit/files/13896668/c3n1.pcap.gz)
;
c1n3.pcap is the configuration of Tx dual antenna and Rx single antenna.

Gi-z, 2024-02-02:
Thanks for providing examples of both configurations, this is really helpful. I've got a Nexus 6P sitting here but the battery is absolutely fried.

Can you confirm the mapping of Tx/Rx is correct with these .pcap samples? c1n3 should contain data with 1 Tx/2 Rx, and c3n1 should contain data with 2 Tx/1 Rx. And so your issue is with the `--csv` option only outputting data from one antenna?

sarielkk, 2024-02-06:
> Thanks for providing examples of both configurations, this is really helpful. I've got a Nexus 6P sitting here but the battery is absolutely fried.
>
> Can you confirm the mapping of Tx/Rx is correct [...]

I can make sure the mapping of this data is correct, the data is collected by myself.
The fact is that c3n1.pcap is 2Tx/1Rx; c1n3.pcap is 1 Tx/2Rx.
My problem is that when I execute `csikit c3n1.pcap`, c3n1 displays 1 Tx/2Rx and c1n3 displays 2 Tx/1 Rx. The numbers of Tx and Rx are exactly the opposite of what they actually are.
I don't know if this has happened to others.