Gi-z/CSIKit #33: IndexError: List of Range

dariabutyrskaya, 2022-11-28:
Hello, I am having similar issues as [this user](https://github.com/Gi-z/CSIKit/issues/26), however i try to run another your work [WiFi-based Human Activity Recognition using Raspberry Pi](https://github.com/Gi-z/ICTAI-2020-Pi4-Activity-Recognition).

The actual issue is when the dataset for standup loaded for 50% it halts and throw this errors.

`Loaded 52.34% for Activity: standup
Loaded 53.27% for Activity: standup
Traceback (most recent call last):
  File "main.py", line 60, in <module>
    runExperiments(config, datasets, resultsFile)
  File "main.py", line 46, in runExperiments
    experiment = Experiment(dataset, config, resultsFile)
  File "/home/jovyan/work/Wifi_activity_recog_pi/activity_recog/ICTAI-2020-Pi4-Activity-Recognition-main/experiment.py", line 45, in __init__
    data = NewPi(config[dataset]["directory"], config[dataset]["config"])
  File "/home/jovyan/work/Wifi_activity_recog_pi/activity_recog/ICTAI-2020-Pi4-Activity-Recognition-main/handlers/newpi.py", line 21, in __init__
    self.initialiseData(directory, config["downsample"])
  File "/home/jovyan/work/Wifi_activity_recog_pi/activity_recog/ICTAI-2020-Pi4-Activity-Recognition-main/handlers/newpi.py", line 123, in initialiseData
    self.x_all, self.y_all = self.loadDataForActivities(directory, saveToFile=True, downsample=downsample)
  File "/home/jovyan/work/Wifi_activity_recog_pi/activity_recog/ICTAI-2020-Pi4-Activity-Recognition-main/handlers/newpi.py", line 112, in loadDataForActivities
    inputArray, outputLabels = self.loadByActivityLabel(directory, activity, activities, saveToFile=saveToFile, windowSize=windowSize, step=step, downsample=downsample)
  File "/home/jovyan/work/Wifi_activity_recog_pi/activity_recog/ICTAI-2020-Pi4-Activity-Recognition-main/handlers/newpi.py", line 90, in loadByActivityLabel
    csiOutput = self.loadFromDat(inputFile, windowSize=windowSize, step=step)
  File "/home/jovyan/work/Wifi_activity_recog_pi/activity_recog/ICTAI-2020-Pi4-Activity-Recognition-main/handlers/newpi.py", line 27, in loadFromDat
    csi_data = reader.read_file(inputFile)
  File "/opt/conda/lib/python3.7/site-packages/CSIKit/reader/readers/read_pcap.py", line 284, in read_file
    data_frames = self.read_frames(self.pcap.frames, scaled, ret_data.bandwidth)
  File "/opt/conda/lib/python3.7/site-packages/CSIKit/reader/readers/read_pcap.py", line 417, in read_frames
    if frames[0].payloadHeader["sequence_no"] == frames[-1].payloadHeader["sequence_no"]:
IndexError: list index out of range
(pi2_env) jovyan@2867d1e59227:~/work/Wifi_activity_recog_pi/activity_recog/ICTAI-2020-Pi4-Activity-Recognition-main$ ` 

My best guess that it is related somehow to CSIKit. Any idea how to solve it? Thanks :)

Gi-z, 2022-11-28:
Hi, this is an issue with CSIKit and I'll hopefully resolve it soon.

In the meantime, this issue can be temporarily resolved by deleting the following files from the `data` directory:
 - standup_sofa_1591623785.pcap
 - standup_sofa_1591623796.pcap

These captures failed during the data collection stage and do not factor into the overall results.
Let me know if this resolves the issue for you 👍

dariabutyrskaya, 2022-11-28:
Yep, that works! Thank you a lot and thanks for your great work 😊

closed 2022-11-28 by dariabutyrskaya