Gi-z/CSIKit #34: Nexmon IndexError: list index out of range

nightball, 2022-12-21:
Hi there!

I am trying to use CSIKit to parse .pcap files from raspberry pi 4B. However, I'm constantly facing an error "IndexError: list index out of range". I used to think there was something wrong with my hardware, but the same problem still occurs when sampling with another device.

I have tried these files: 
[output.pcap](https://drive.google.com/file/d/1bk2TPRGmoRN6Z8jY0SQQ5l391mGGaUPW/view?usp=sharing), [output2.pcap](https://drive.google.com/file/d/1M10wEvQ9N7Q0a0vxNKzFly4JqRrG8sHz/view?usp=sharing) or [outputhome.pcap](https://drive.google.com/file/d/1kk3DN7JbI8UaPf5yLjDmEB2RX8FC2QFR/view?usp=sharing)

When I try running the "Using CSI Matrices" sample code you provide
```
from CSIKit.filters.passband import lowpass
from CSIKit.filters.statistical import running_mean
from CSIKit.util.filters import hampel

from CSIKit.reader import get_reader
from CSIKit.tools.batch_graph import BatchGraph
from CSIKit.util import csitools

import numpy as np

my_reader = get_reader(r"C:\Users\HoChunHi\OneDrive\Desktop\Work\FYP_python\output2.pcap")
csi_data = my_reader.read_file(r"C:\Users\HoChunHi\OneDrive\Desktop\Work\FYP_python\output2.pcap", scaled=True)
csi_matrix, no_frames, no_subcarriers = csitools.get_CSI(csi_data)

# CSI matrix is now returned as (no_frames, no_subcarriers, no_rx_ant, no_tx_ant).
# First we'll select the first Rx/Tx antenna pairing.
csi_matrix_first = csi_matrix[:, :, 0, 0]
# Then we'll squeeze it to remove the singleton dimensions.
csi_matrix_squeezed = np.squeeze(csi_matrix_first)

# This example assumes CSI data is sampled at ~100Hz.
# In this example, we apply (sequentially):
#  - a lowpass filter to isolate frequencies below 10Hz (order = 5)
#  - a hampel filter to reduce high frequency noise (window size = 10, significance = 3)
#  - a running mean filter for smoothing (window size = 10)

for x in range(no_frames):
  csi_matrix_squeezed[x] = lowpass(csi_matrix_squeezed[x], 10, 100, 5)
  csi_matrix_squeezed[x] = hampel(csi_matrix_squeezed[x], 10, 3)
  csi_matrix_squeezed[x] = running_mean(csi_matrix_squeezed[x], 10)

BatchGraph.plot_heatmap(csi_matrix_squeezed, csi_data.timestamps)
```
then it return the error:

```
Traceback (most recent call last):
  File "c:\Users\HoChunHi\OneDrive\Desktop\Work\FYP_python\tempCodeRunnerFile.py", line 12, in <module>
    csi_data = my_reader.read_file(r"C:\Users\HoChunHi\OneDrive\Desktop\Work\FYP_python\output2.pcap", scaled=True)
  File "C:\Users\HoChunHi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\CSIKit\reader\readers\read_pcap.py", line 284, in read_file
    data_frames = self.read_frames(self.pcap.frames, scaled, ret_data.bandwidth)
  File "C:\Users\HoChunHi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\CSIKit\reader\readers\read_pcap.py", line 417, in read_frames
    if frames[0].payloadHeader["sequence_no"] == frames[-1].payloadHeader["sequence_no"]:
IndexError: list index out of range

```
Any idea how to solve it ? Hope you can offer me some suggestions!!

Gi-z, 2022-12-21:
Hi, I just confirmed this issue on my end. I've pushed a fix for this and confirmed it works with each of the files you provided while retaining backwards compatibility.

You can try installing the new version with the following command: 
`pip install --upgrade git+https://github.com/Gi-z/CSIKit.git`

Please try it and let me know if it works for you!

nightball, 2022-12-22:
It seems work for me too !! Thank you so much for your help

closed 2022-12-22 by nightball