Gi-z/CSIKit #35: TypeError: NEXBeamformReader.read_file() got an unexpected keyword argument 'filter_mac'

goebelko, 2022-12-22:
Hi there!

Thanks a lot for sharing this useful tool！！

I'm trying to use CSIKit to parse .pcap files from raspberry pi 4b. Yet,  I'm getting the TypeError: NEXBeamformReader.read_file() got an unexpected keyword argument 'filter_mac'. 
Example sample of files: 
[https://drive.google.com/file/d/1n0B2uxjvbpMWpXHOOPRzgtCmpQrq8Rum/view?usp=sharing](url)

[image]

Any suggestions you can offer would be hugely appreciated - thank you again !

Gi-z, 2022-12-22:
Hi, can you confirm you're running the latest version of CSIKit? This was an issue with older versions. Should be working in 2.4. You can install it with the following command: `pip install --upgrade csikit==2.4`. Let me know how you get on 👍

goebelko, 2022-12-23:
hi i try to run this command: python -m pip install 
[image]
and this is the latest version of CSIKit(2.4).

Gi-z, 2022-12-23:
Ah, right. Could you try this command instead? This'll install the latest pre-release build from this repository.

pip install --upgrade git+https://github.com/Gi-z/CSIKit.git

CristianTuretta, 2023-01-02:
hi, same issue... both commands suggested by @Gi-z didn't work

Gi-z, 2023-01-02:
> hi, same issue... both commands suggested by @Gi-z didn't work

Hi, I just tried this out on a fresh machine to confirm it works. You can see I checked CSIKit wasn't installed, and then I installed it using the command provided. Once installed, I confirmed function with 3 different files including the one provided by @goebelko above.
https://gist.github.com/Gi-z/16997711e91d8a58be75f74e0a8c55b5

The issue likely stems from having the current public version of CSIKit installed which pip will not upgrade to the pre-release build as the version numbers are the same. If you force uninstall CSIKit first using `pip uninstall csikit` and then reinstall with `pip install --upgrade git+https://github.com/Gi-z/CSIKit.git`, this should resolve the issue. Please let me know if this doesn't solve your problem.

CristianTuretta, 2023-01-02:
Thank you @Gi-z now it works 👍

Gi-z, 2024-02-13:
Resolved in newest build.

closed 2024-02-13 by Gi-z