Gi-z/CSIKit #39: how can i solve this problem? (empty part of data)

rack570, 2023-05-28:
Hello, I'm new to nexmon extractor & CSIkit. I was able to extract the pcap file correctly according to the instructions. But I have a few questions. I tried to visualize the data using Gi-z's CSIKit, but there is a part in the middle where there is no data (white part). How can I solve this phenomenon? And there's one more thing. Visualizing the data extracts the same 2000 packets, each of which has a different part about time on the x-axis. Is there a way to time each extraction data?

channel 6 bandwidth 20MHz (there is no last data)
[image]

channel 44 bandwidth 80MHz (data is empty between.)
[image]

----------------------------------------------------
from CSIKit.filters.passband import lowpass
from CSIKit.filters.statistical import running_mean
from CSIKit.util.filters import hampel

from CSIKit.reader import get_reader
from CSIKit.tools.batch_graph import BatchGraph
from CSIKit.util import csitools

import numpy as np

my_reader = get_reader("path pcap file")
csi_data = my_reader.read_file("path pcap file", scaled=True)
csi_matrix, no_frames, no_subcarriers = csitools.get_CSI(csi_data, metric="amplitude")

csi_matrix_first = csi_matrix[:, :, 0, 0]
csi_matrix_squeezed = np.squeeze(csi_matrix_first)

for x in range(no_frames):
  csi_matrix_squeezed[x] = lowpass(csi_matrix_squeezed[x], 10, 100, 5)
  csi_matrix_squeezed[x] = hampel(csi_matrix_squeezed[x], 10, 3)
  csi_matrix_squeezed[x] = running_mean(csi_matrix_squeezed[x], 10)

BatchGraph.plot_heatmap(csi_matrix_squeezed, csi_data.timestamps)

Gi-z, 2023-05-28:
Hi, I think there's a typo in the example. Could you send me the pcap file you're using and I'll try fixing it?

rack570, 2023-05-28:
[output(2000packets).zip](https://github.com/Gi-z/CSIKit/files/11584573/output.2000packets.zip)
Thanks for your kindness! I saw your paper recently. "wifi based human activity recognition using raspberry pi" Thank you very much for your reply.

Gi-z, 2023-05-28:
Hi, I've included an updated example with some additional comments.

There are some important things to note:

- If you are performing any kind of signal processing operation, you must know the sample rate of data. If the CSI data contains valid timestamps, then you can use CSIKit's command line interface to check the average sample rate in a given file. Example command: `csikit ch44bw80.pcap`. Example output:
```console
C:\Users\Giz\Downloads\output.2000packets>csikit ch44bw80.pcap
Hardware: Broadcom BCM43455c0
Backend: Nexmon CSI
Bandwidth: 80MHz
Antenna Configuration: 1 Rx, 1 Tx
Frame Count: 1152
Subcarrier Count: 256
Length: 108.24s
Average Sample Rate: 10.60Hz
Average RSSI: -65.8dBm
CSI Shape: (1152, 256, 1, 1)
```
- You must also know why you are applying each step of the processing sequence. In the original example, first a butterworth lowpass filter is applied. This is performed with cutoff of 10Hz, meaning that any signals above 10Hz in your data will be filtered out. This operation can be useful in filtering out high frequency noise, but may distort other signals in the lower bound. A lowpass (or any type of passband) filter requires you to know the sample rate of your data, and so running it with the default 100Hz parameter provided in the example will not produce a meaningful result. Given your data is only 10Hz, you will be unable to perform any meaningful processing with passband filters, so I've disabled it in the new example I've provided.
- Similarly, the other operations are being run with a window size of 10, which in your case would mean a 1s window. This probably isn't what you're going for, and so even in low sample rate scenarios you'd probably be interested in .5s windows. I've changed the window size of the other 2 filters to 5 samples in the new example I've provided.

```python
from CSIKit.filters.passband import lowpass
from CSIKit.filters.statistical import running_mean
from CSIKit.util.filters import hampel

from CSIKit.reader import get_reader
from CSIKit.tools.batch_graph import BatchGraph
from CSIKit.util import csitools

import numpy as np

my_reader = get_reader("ch44bw80.pcap")
csi_data = my_reader.read_file("ch44bw80.pcap", scaled=True)
csi_matrix, no_frames, no_subcarriers = csitools.get_CSI(csi_data, metric="amplitude")

csi_matrix_first_antenna = csi_matrix[:, :, 0, 0]
csi_matrix_squeezed = np.squeeze(csi_matrix_first_antenna)

# Data is now in format (no_frames, no_subcarriers)
# We need to iterate through each subcarrier stream and apply our processing functions.
# An easy way to do that is to transpose the matrix.
csi_matrix_transposed = np.transpose(csi_matrix_squeezed)

# Now the data is in the format (no_subcarriers, no_frames)

for x in range(no_subcarriers):
    # csi_matrix_transposed[x] = lowpass(csi_matrix_transposed[x], 5, 10, 3)
    csi_matrix_transposed[x] = hampel(csi_matrix_transposed[x], 5, 3)
    csi_matrix_transposed[x] = running_mean(csi_matrix_transposed[x], 5)

# We can then transpose the matrix again so we can arrange our axes.
csi_matrix_disp = np.transpose(csi_matrix_transposed)

BatchGraph.plot_heatmap(csi_matrix_disp, csi_data.timestamps)
```

[image]

rack570, 2023-05-30:
Thanks very much for your help. But I have a few more questions..

1. I tried to open the example pcap file using the command csikit in the kernel and look at the sample rate, but the error "invalid syntax" appears.

2. So is this csikit only available if the sample rate is 100Hz? Can I change the sample rate directly from router settings or nexmon extractor?

I'm a person who's trying to study OCC (Occupant Centric Control). I was looking for a way to do real-life sensing with device-free, but I'm getting to know CSI belatedly, so I'm sorry that I have a lot of questions. I'm trying to use python, but I'm using python analysis tools because it seems to be used a lot for you. Thank you in advance for your reply.

Gi-z, 2023-05-30:
1. If you follow the syntax used in my example above, there will not be an error. You simply use the command `csikit yourfilename.pcap`. If there is an error when you do this, please post a screenshot here.
2. CSIKit supports any given sample rate. The `lowpass` filter used in the example above has a positional parameter `fs` in which you must supply the sample rate of your input signal. However it will not be very effective if your data is only at a 10Hz sample rate (as mentioned before).

rack570, 2023-06-01:
pip install --upgrade git+https://github.com/Gi-z/CSIKit.git

- When I uninstalled csikit and changed it to this command after seeing another issue, problem 1 was solved. Thank you

- What I am more curious about is the sample rate.. Is the higher the sample rate better data? As a result of checking with the csikit command, the data I selected shows that the csi data extracted from the 80mhz bandwidth has an average sample rates of 10hz and the data extracted from the 20mhz bandwidth is 0hz. I looked up other issues and found that there are ones with sample rates of 0hz. Is this wrong data? How do I extract data at a high sampling rate of 100hz like your research?

I always appreciate your help.

rack570, 2023-06-01:
example pcap files (average sample rate = 0hz)
[csiexample.zip](https://github.com/Gi-z/CSIKit/files/11623306/csiexample.zip)

rack570, 2023-06-05:
I'm sorry I kept asking you questions.
I think I can understand the story you said as I study it more and more.
After checking the sampling rate for each data, we are observing the data by adjusting the window size accordingly. I still don't know why my sampling rate is 0hz... I'd like to ask another question. If the sampling rate is as low as mine, is it right not to use the low pass filter? Even if the window size and cutoff of the low pass filter are lowered, it continues to disconnect during the plot.. And how can I match each data with the time on the x-axis? Comparison is difficult due to different time axes. If it's hard, is there a way to express the number of packets on the x-axis instead of time like the plot method in the seemoo lab? As far as I know, nexmon extractor can only be extracted by the number of packets. I'm always waiting for your answer. Thank you.

The number of packets on the x-axis like this picture
[image]

liuchengzhi314159, 2024-10-12:
> 你好，我是nexmon提取器和CSIkit的新手。我能够根据说明正确提取 pcap 文件。但我有几个问题。我尝试使用 Gi-z 的 CSIKit 可视化数据，但中间有一个部分没有数据（白色部分）。我该如何解决这个现象呢？还有一件事。可视化数据会提取相同的 2000 个数据包，每个数据包在 x 轴上都有关于时间的不同部分。有没有办法为每个提取数据计时？
> 
> 通道 6 带宽 20MHz（无最后数据） ![Figure 2023-05-28 210636](https://private-user-images.githubusercontent.com/131440518/241558771-bf734269-ae9a-4435-bdda-f3afb0588c16.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3Mzc1NzUsIm5iZiI6MTcyODczNzI3NSwicGF0aCI6Ii8xMzE0NDA1MTgvMjQxNTU4NzcxLWJmNzM0MjY5LWFlOWEtNDQzNS1iZGRhLWYzYWZiMDU4OGMxNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxMlQxMjQ3NTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kZGJmOWY1ZmRiNzQxZmEwNTZjYjc5MDk2NTE5ZDAxOWI5MmFjNDc5Yzg0N2U2MGZlYTgwZGI1NTRjOTg4MDU1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.spN26i1WXe1AWm0Q8NM2LD0OK-OXxMtRIUPuNU6jH78)
> 
> 通道 44 带宽 80MHz（数据为空。 ![ch44bw80](https://private-user-images.githubusercontent.com/131440518/241558832-97f1e566-7962-4af8-8d53-fbf8c313e0cb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3Mzc1NzUsIm5iZiI6MTcyODczNzI3NSwicGF0aCI6Ii8xMzE0NDA1MTgvMjQxNTU4ODMyLTk3ZjFlNTY2LTc5NjItNGFmOC04ZDUzLWZiZjhjMzEzZTBjYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxMlQxMjQ3NTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iYzQ0NjdmOTJmYmZmNmI4MmNlMjhmN2NiOTBlYjRhZmJmMTQzMTlhNzhkNmE0YzYwNjViZjdjMmQwNTU1OTlkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.eYe4aETAyW_vGzLJi1A4Gis-PgVqAnQmfU52340_bq4)
> 
> from CSIKit.filters.passband import lowpass from CSIKit.filters.statistical import running_mean from CSIKit.util.filters import hampel
> 
> from CSIKit.reader import get_reader from CSIKit.tools.batch_graph import BatchGraph from CSIKit.util import csitools
> 
> 将 numpy 导入为 NP
> 
> my_reader = get_reader（“路径 pcap 文件”） csi_data = my_reader.read_file（“路径 pcap 文件”， scaled=True） csi_matrix， no_frames， no_subcarriers = csitools.get_CSI（csi_data， metric=“amplitude”）
> 
> csi_matrix_first = csi_matrix[：， ：， 0， 0] csi_matrix_squeezed = np.squeeze（csi_matrix_first）
> 
> 对于范围 （no_frames） 中的 x： csi_matrix_squeezed[x] = 低通（csi_matrix_squeezed[x]， 10， 100， 5） csi_matrix_squeezed[x] = 汉佩尔（csi_matrix_squeezed[x]， 10， 3） csi_matrix_squeezed[x] = running_mean（csi_matrix_squeezed[x]， 10）
> 
> BatchGraph.plot_heatmap（csi_matrix_squeezed、csi_data.timestamps）

Hello, I would like to ask the following what I should do or what commands I should enter to achieve my purpose: I would like to use nexmon on a raspberry pi 4b to get the csi information and save it locally for further work.
System version: linux kernel 6.1 has changed supported drivers