Gi-z/CSIKit #41: Graph Dimension Mismatched

Phon1209, 2023-07-02:
I'm new to this project and know very little about Nexmon and CSIKit.

When I tried using CSIKit to plot a graph using the command line provided as an example on my own pcap data, this error occurred:
`ValueError: x and y must have same first dimension, but have shapes (250,) and (256, 1, 1)`

Here is the info I got from using `--info`:
```
Backend: Nexmon CSI
Bandwidth: 80MHz
Antenna Configuration: 1 Rx, 1 Tx
Frame Count: 250
Subcarrier Count: 256
Length: 19.61s
Average Sample Rate: 12.70Hz
Average RSSI: -36.4dBm
CSI Shape: (250, 256, 1, 1)
```

Did I miss something or is something wrong with the data?

yqqhi, 2023-09-14:
Did you solve this problem, I also encountered the same situation and did not know how to deal with it

Gi-z, 2024-02-02:
Hopefully you've managed to get the later commits of CSIKit installed which should've fixed this problem. A new version of CSIKit (v2.5) has just been published on PyPi which will allow you to install it with `pip install --upgrade csikit==2.5`. Let me know if you're still having issues! Closing this for now.

closed 2024-02-02 by Gi-z