Gi-z/CSIKit #47: csikit file_name.pcap error, how to resolve it?csikit file_name.pcap error, how to resolve it?

sherlock-L-L, 2023-12-21:
hi, thank you for your csikit.
[image]

sherlock-L-L, 2023-12-21:
I still get an error when I use this command（pip install --upgrade git+https://github.com/Gi-z/CSIKit.git
[image]
）

doge0823, 2023-12-29:
Hello, now I have the same problem, may I ask you to solve this problem?

sherlock-L-L, 2024-01-04:
> Hello, now I have the same problem, may I ask you to solve this problem?

I'm sorry, I still haven't resolved this issue

doge0823, 2024-01-04:
Thanks for your reply, I have successfully implemented it in python. I repeated the two sentences of the author's command two or three times, and then suddenly found that it could be executed.
```
pip uninstall csikit
pip install --upgrade git+https://github.com/Gi-z/CSIKit.git
```

sherlock-L-L, 2024-01-04:
> 谢谢你的回复，我已经用python成功实现了。我把开发者提供的两条命令重复了两三遍，然后突然发现可以执行了。
> 
> ```
> pip uninstall csikit
> pip install --upgrade git+https://github.com/Gi-z/CSIKit.git
> ```

I repeated it several times, but it didn't work.

doge0823, 2024-01-04:
May be related to the slow Github network connection, python also shows " fatal: unable to access https:// github.com/gi-z/csikit.git /' ", I suggest that you can try to use VPN software to change the network environment.

Gi-z, 2024-02-02:
Hi all, sorry for the lack of communication. I've pushed the latest (working!) version of CSIKit (v2.5) to PyPi, so you should be able to update now without using the GitHub option in pip. Let me know if you have any other issues with this and I'd be happy to clear them up. Closing this for now.

closed 2024-02-02 by Gi-z