Gi-z/CSIKit #59: OverflowError: Python integer 2806 out of bounds for int8

BXLeohim, 2024-08-17:
Thank you for your powerful toolkit, which helped me take the first step in CSI learning. I am very grateful for your help.

I am a beginner in CSI and currently learning about phase unwrapping and phase adjustment. I have encountered a problem at the moment, and I would greatly appreciate it if you could help me.

When I tried to read data from the Widar3.0 dataset, the program encountered the following error:
`Traceback (most recent call last):
  File "c:\Users\BX\Desktop\独居老人\CSI_read_python\csi_read1.py", line 19, in <module>
    csi_data = my_reader.read_file(file_path, scaled=1)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\BX\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\CSIKit\reader\readers\read_bfee.py", line 180, in read_file
    csi_matrix = IWLBeamformReader.read_bfee(data_block, n_tx, n_rx, expected_length, perm, ret_data.expected_frames)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\BX\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\CSIKit\reader\readers\read_bfee.py", line 78, in read_bfee
    real = np.int8(real)
           ^^^^^^^^^^^^^
OverflowError: Python integer 2806 out of bounds for int8`

I don't know where the problem lies, whether it's with the dataset or my tool usage.

Thank you so much!

maple-42, 2024-09-05:
Hi.
I am getting the same error, did you solve this problem?

Best regards.

Gi-z, 2024-09-05:
Hi, looks like `np.int8` getting deprecated might be causing this. I'm back from vacation in a week, but in the meantime you should try using an older version of Python (3.8 or 3.9). Thanks for reporting and I'll make sure it gets sorted once I'm back.

I understand you're using the Widar3.0 dataset, but could you provide an example file which causes this error? Thanks.

maple-42, 2024-09-06:
Thanks for the reply.
I am sorry I only saw the error and not the dataset. I am using the nexmon dataset.
I will also try to use an older version of Python.
I will also attach a sample file.
Thank you very much.

[2.zip](https://github.com/user-attachments/files/16903362/2.zip)

BXLeohim, 2024-09-06:
Thank you for your reply
I changed Python to 3.8 according to your method, and there were no issues with the program

I will also attach an example file
Thank you again

[user6-1-1-1-1-r1.zip](https://github.com/user-attachments/files/16904003/user6-1-1-1-1-r1.zip)

maple-42, 2024-09-09:
Sorry for the delay in replying.
I will try it with Python 3.8.
Thanks also for the example file.

maple-42, 2024-09-12:
The result of the execution is shown and illustrated in this figure.
I am not sure if I got the correct result, but I was able to run it for the time being.

Thank you very much.
[image]

Gi-z, 2024-09-16:
Hi all, I've pushed a commit to handle this issue. A new CSIKit version will be released shortly.

closed 2024-09-16 by Gi-z